upb-lea/ElectricGrid.jl #4: Examples

MarvinMeyer, 2022-02-08:
This issue covers the creation of examples for all functions of the package. The examples should be well documented and easy to follow.

TODO:
- Python Notebooks
- [ ] NodeConstructor

- Python Files
- [ ] NodeConstructor

Webbah, 2022-09-30:
Use it in general for example or close?

SeptimusBoshoff, 2023-02-07:
Classical Controllers - Jupyter Notebooks

- [ ] Swing mode + Static Load
- [ ] Droop mode + Static Load
- [ ] Droop mode + Controllable Load
- [ ] VSG mode + Static Load + Controllable Load

Webbah, 2023-02-24:
- [ ] WebIO not detected-error.
How can we solve that? Even if the plots show up, reopening the notebook results in this error again

Webbah, 2023-08-25:
@janstenner and @SeptimusBoshoff, are the mentiones notebooks in progress or is it a wont fix?
If the latter, please mention or close the issue

closed 2023-09-22 by MarvinMeyer