upb-lea/ElectricGrid.jl #15: First RL Experiments

janstenner, 2022-05-25:
Implement first experiments with the julia implementation of the RL environment and a DDPG agent

Webbah, 2022-05-27:
1. Simple example with 1 source, supplying 1 load via 1 cable
-  Load: stateless -> R
- Goal: supply reference power P* to the load
- assume all states to be measurable and provided to the agent
- How is the reward function designed? 

2. Increase the power grid example
- More loads and inverters but still 1 agent
- Do we need more information in the reward function for example to share the load across the sources fairly?
- How is the reward function designed?

Webbah, 2022-09-30:
Still in progress?

VikasChidananda, 2022-10-26:
Part 1 is done with one source and one load. used reward function - power optimization with voltage constraint
Part 2 is not done yet.

Webbah, 2022-12-16:
still needed? Close the issue or is still work in progress here?

VikasChidananda, 2022-12-16:
- [ ]  Simple example with 1 source, supplying 1 load via 1 cable

Load: stateless -> R
Goal: supply reference power P* to the load
assume all states to be measurable and provided to the agent
How is the reward function designed?

- [ ] Increase the power grid example

More loads and inverters but still 1 agent
Do we need more information in the reward function for example to share the load across the sources fairly?
How is the reward function designed?

//Needed for the second example.

VikasChidananda, 2022-12-20:
- [ ] [reward normalizer](https://github.com/JuliaReinforcementLearning/ReinforcementLearning.jl/blob/2e1de3e5b6b8224f50b3d11bba7e1d2d72c6ef7c/src/ReinforcementLearningZoo/src/utils/reward_normalizer.jl#L38-L50)

Webbah, 2023-05-03:
@VikasChidananda  is this still in progess/under development?
If not, please close

closed 2023-05-04 by VikasChidananda