upb-lea/ElectricGrid.jl #17: Nodeconstructor Interface

Webbah, 2022-06-02:
Based on the in #3 developed Nodeconstructor which is able to set up a specific power grid structure with different sources, loads and connections in the state space, a popper **Interface** is needed to access specific states/information.
Theses information shall be provided for example to:
 - classical controllers 
 - data-driven controllers
 - GUI
 - Dashboard

Part of the information need to be:
 - type of component (source, load, cable)
 - limits of the component
 - additional information (V_dc, length,...?)
 - ...?

The result is intended as a uniform interface and shall enable the user to choose different kind of control approaches for all defined components.

Webbah, 2022-09-30:
### Update/still toDo

Most of the work is already implemented in the parameter dict of the node constructor.
This dict has to be double checked since when not all parameters e.g. of a filter are defined non of them are used and just drawn all random. 
Additionally it has to be checked that all needed information is available like 

- [ ] V_dc
- [ ] Limits (related to #24 )
- [ ]  active/reactive power points
- [ ] power factor (?)
- [ ] voltage peak set point
- [ ] phase angle
- [ ] more needed e.g. for Classic Controllers @SeptimusBoshoff ?

Some of the labeling might be linked to #22

janstenner, 2022-12-01:
Found two issues:

1. When no parameters are given and the num_source and num_load parameters do not match CM size, line 54 wants to throw an error:
`if size(CM)[1] != tot_ele
            throw("Expect the number of elements in the node to match the specified structure in the CM, but got $tot_ele and $(size(A)[1])")
        end`
The Problem here is: A does not exist yet.

2. If I only provide sources with only "pwr" and "vdc" set, it runs into line 273
`if source["fltr"] == "LC"`
There it expects "filtr" to be set, which isn't

SeptimusBoshoff, 2022-12-20:
1. Should the user be able to specify a CM without any passive loads? For example, only a network with sources.
2. Rename PV (photovoltaic) in source type to Solar to avoid confusion with PV (active power and voltage magnitude) in control mode. Preference is given to the control mode, because "Solar" is shorter than "Active power and voltage magnitude".

closed 2023-05-03 by Webbah