upb-lea/ElectricGrid.jl #22: Enviroment extensions

OliverSchw, 2022-07-05:
- [ ] create history
- [ ] function to plot states (consider 3 phases)
- [ ] output specific states for selected inverter etc.

SeptimusBoshoff, 2022-07-05:
I have some functionality already developed in the branch: /feature_19_Classical_Control/pre_investigations/julia/scripts. There should be 3 julia files.
1. Classical_Control_Plots - should be the one that is most applicable.
2. DC_to_AC_Converter_Control_08 - example script with 2 sources and 1 load (well actually 2 at the same node). 
3. Classical_Control.jl - creates Source Control and Environment struct, also defines Measurements functions

At the bottom of the example script are all the function calls to the plots.

Other comments:
1. The handling of the A, B, C, D three phase matrices are done by the function Two_Source_One_Load in Classical_Control.jl

Webbah, 2022-09-30:
Most work is implemented using the data_hook and DataFrames (https://github.com/upb-lea/dare/blob/develop/src/data_hook.jl)
- Based on the in #17 defined labels and ids (env.state_ids, using get_state/action_ids(env.nc)) the data can be selected to store with the hook in the dataframe.

That data can be used in a plotting function (https://github.com/upb-lea/dare/blob/develop/src/plotting.jl) which uses the same labeling to plot selected data using plotly.

Some data-information and plotting functionality still has to be implemented e.g. to show all results of the Classic controllers implemented in #19 
For Example:

- [ ] Power active/reactive
- [ ] dq0 Qantities
- [ ] ....?
@SeptimusBoshoff feel free to add points to this collection and edit this comment directly

SeptimusBoshoff, 2023-01-17:
How should the featurize be implemented? Seems like there is a splitting of states between classical and RL, which complicates things - maybe. @janstenner

Webbah, 2023-05-03:
featureized is refactored, rest in this issue should be done