upb-lea/ElectricGrid.jl #23: WP 2 - Modeling of a PV plant with variable DC-Link voltage

MarvinMeyer, 2022-09-06:
This issue deals with the modeling of a PV system. Based on irradiation and temperature curves, a variable output voltage should be obtained from the physical equations, depending on the filter current.

The implementation in Matlab ([Source](https://de.mathworks.com/help/physmod/sps/powersys/ref/pvarray.html)), based on the paper ([Source](https://www.sciencedirect.com/science/article/pii/S2090997714000182)), serves as a reference for the model.

In the first step, a single-phase implementation is to be created, which can later also be used for the three-phase case with the use of an inverter.

Webbah, 2022-09-29:
### General implementation of v_dc as function, callable in the env.step:

First ideas/brainstroming/keep in mind...:

- Compareable to the featureize function we could write a function that is called depending on the topology of the source (PV, Batt, windturbine,...).
- The topology could be labeled in the parameter dict. 
   - By default labeled as "ideal" assuming infinite source -> constant v_dc
- Example "PV": parameters of the PV-plant given in the parameter dict and used when the function is applied in the env.step

MarvinMeyer, 2022-12-19:
Current status:

- PV systems are ready, but still need to be tested
- Batteries are in progress

MarvinMeyer, 2023-04-21:
- [ ] Model internal resistance for loss calculation

MarvinMeyer, 2023-08-11:
Create a structure containing the individual DC link voltages. In addition, the calculation of the stresses should be delayed by one time step.

MarvinMeyer, 2023-09-15:
Merged branch into develop.

closed 2023-09-15 by MarvinMeyer