upb-lea/ElectricGrid.jl #27: Power ratings

Webbah, 2022-12-01:
Seems like some calulations in the NC are assuming a 3-phase grid for example
https://github.com/upb-lea/dare/blob/a5d744c31a2adbad795cb55d9c2162221ce689aa/src/nodeconstructor.jl#L244

That `3` should be changed to parameter["grid"]["phase"]
Where else we have hard coded assumtions like 3-phase grid? -> double check

Is the total power rating already checked? So that pwr_sources >= pwr_loads?
What happens if that is equal? In the cable ratings #24 we might run into problems in the optimization.
We should add a "power-level" depending offset to consider worst case power losses across the cables

Webbah, 2023-01-12:
#24 take parameter["grid"]["phase"] into account during caculation/cable ratings

MarvinMeyer, 2023-02-10:
"Random" sampling of the Sources and Loads results in highly inbalanced powerratios. Thus the PFE can't be solved, resulting in an error.

- [ ]  Verified?

Solution would be to set the power in the grid fix (i.e. grid_pwr = num_sources * pwr) and make sure the loads are sampled in a way that it fits to the grid_power (i.e. load_pwr = grid_pwr/num_loads)

- [ ] Done?

SeptimusBoshoff, 2023-02-22:
the current and voltage limits of the loads

- [ ] done?

The voltage limit is easy. The difficulty comes in for loads with inductors. What needs to be done is the complex power absorbed by just the inductor should be calculated. S = V I* is the equation. The voltage will be the minimum voltage, something like 0.9 p.u., such that the current limit is maximised.

https://github.com/upb-lea/dare/blob/3ad14d4c44f26d53ff4ab53d7b24f7be028d8645/src/nodeconstructor.jl#L756

closed 2023-03-20 by Webbah