upb-lea/ElectricGrid.jl #31: Documentation

Webbah, 2023-01-27:
## User Documentation 
- Introduction to what is Dare?
- Overview functionality
- Tutorials 
  -  Install
  - Examples
  - Videos?

## API Documentation
- All (?) functions to be documented with appropriate syntax

## Tool(s)
- https://documenter.juliadocs.org/stable/

Webbah, 2023-02-08:
- variable names: snake_case (small+split by _)

- function names: PascalCase 

- file names: (do not change old functions now!) snake_case

Webbah, 2023-02-08:
- [ ] keep github readme up to date

Webbah, 2023-02-17:
- [ ] Name for the package
- [ ] Name for the env

SeptimusBoshoff, 2023-02-23:
for every source file

- [ ] Classical Controllers
- [ ] nodeconstructor
- [ ] environment
- [ ] power system theory
- [ ] dare wrapper

VikasChidananda, 2023-02-24:
Julia package naming guidelines: https://pkgdocs.julialang.org/v1/creating-packages/

janstenner, 2023-03-16:
I did merge a first version of the documentation with the commit c33e333405fb8d7c1db8b8806348457364f11ba3

Webbah, 2023-05-03:
User guide available - to be kept up to date with new features

closed 2023-05-03 by Webbah