upb-lea/ElectricGrid.jl #44: JOSS paper

wallscheid, 2023-03-23:


Webbah, 2023-03-31:
First Version ready to go.

review plan in suggested order (please note when done and inform next in list):

- [ ] @Webbah 
- [ ] @MarvinMeyer 
- [ ] @SeptimusBoshoff 
- [ ] @janstenner 
- [ ] @VikasChidananda 
- [ ] @SebastianPeitz 
- [ ] @wallscheid

Webbah, 2023-07-26:
Compressed todos from both reviews:

#### Paper
- More background why RL? 
- [ ]  RL Paper Jan

#### Checklist
Refers to Paper from:
- [ ] Statement of need missing  (in paper and documentation)-> rename: Background on electric grids and their control
  - [ ] How does the software compares with state-of-the-art -> Paper Marvin
  - [ ]  Target Audience missing -> we have an own section for that. shift?
 
- [ ]  State of field only 1 reference (-> to Julia package):
   - What is the value-add of Julia compared to python/matlab everyone would use? -> Paper Marvin
   - How does DARE compares to PowerSimulationDynamics.jl -> OS results on Sciebo

#### Documentation
- [ ]  Document all parameters (X_theat?) 
- [ ] correct installation

#### Future enhancements

- change parameter dict to structs -> new enhancement issue?

closed 2023-09-22 by MarvinMeyer