upb-lea/ElectricGrid.jl #45: Exteneded Unit Tests

VikasChidananda, 2023-03-24:
Power flow equations
- [ ] Test for Vrms = 230 V, 100 V to make sure it works for different voltages
- [ ] Add in loads and try different topologies.

VikasChidananda, 2023-04-20:
```Julia
┌ Warning: Power flow equation not solveable! Maybe parameter setting invalid.
│                   Default values are used for cable parameters
└ @ ElectricGrid /data/cvikas/Projects/dare/src/power_system_theory.jl:1186
[ Info: Normalization is done based on the defined parameter limits.
[ Info: Time simulation run time: 0.8 [s] ~> 8001 steps
[ Info: 2 'classically' controlled sources have been initialised.
[ Info: 2 sources have been set up in Voltage mode.
[ Info: 2 sources have automatically calculated proportional and integral gains for their voltage control loops.
[ Info: 2 sources have automatically calculated proportional and integral gains for their current control loops.
p_test = 165.04976255697363
q_test = -19737.968426855794
norm(p_test, q_test) = 165.04976255697363
3 * sum(ElectricGrid.P_source) = 0.0
3 * sum(ElectricGrid.Q_source) = 0.0
power balance: no load: Test Failed at /upb/users/c/cvikas/profiles/unix/cs/Projects/dare/test/power_flow_test_P_Q.jl:176
  Expression: p_test - 3 * sum(ElectricGrid.P_source) < 10.0
   Evaluated: 165.04976255697363 < 10.0
```

Cable layout fails for no - load test.

Webbah, 2023-08-25:
Are the tests mentioned above implemented meanwhile?

closed 2023-09-22 by MarvinMeyer