upb-lea/ElectricGrid.jl #97: Real Time Performance Tests

janstenner, 2023-04-26:
We want to investigate what factors are affecting our performance in which amount and then try to make it faster!

- [ ] create performance benchmarks
- [ ] optimize DataHook
- [ ] optimize Classical Controllers

VikasChidananda, 2023-04-26:
Some specification on performance benchmarks:
- run the benchmarks on CPU and GPU
- Profiling to find bottleneck (if any)
- Metrics for comparison: timesteps / unit time,  updates / unit time 
- vary NN architecture linearly with increasing nodes(N)  [learnable parameters M = AN + b]

- [ ] Open loop simulation (without control)
- [ ] Closed loop simulation (with reinforcement learning / classical controls?)

VikasChidananda, 2023-05-05:
Ideas for configuring `num_sources` and `num_loads`:
- [ ] `num_sources` = `num_loads`
- [ ] set of all (`num_source_i`, `num_loads_j`) such that `num_source_i` + `num_loads_j` = `num_nodes` 
- [ ] stochastically choose `num_source` such that 'num_source' > `num_loads` and `num_source` + `num_loads` = `num_nodes`

VikasChidananda, 2023-05-05:
[Reproducibility pitfalls](https://juliaci.github.io/BenchmarkTools.jl/dev/linuxtips/)

Webbah, 2023-08-25:
the controllers we went through together.
How about the rest?

closed 2023-09-22 by MarvinMeyer