upb-lea/ElectricGrid.jl #113: Editorial comment on JOSS submission

osorensen, 2023-08-27:
The sentence highlighted below does not seem complete. What do you mean by "tools like (Lara et al., 2023)"?

[image]

https://github.com/openjournals/joss-reviews/issues/5616

osorensen, 2023-08-27:
If you mean "tools like the one presented in Lara et al. (2023)", then please rewrite to that. If you need it, note that you can create references without paranetheses with `@key`, i.e., dropping the square brackets.

Webbah, 2023-08-29:
Yes, that was our intention and I am with you that the sentence is a bit "short". 
Thank you for the suggestions, changes have been updated

closed 2023-08-29 by Webbah