upb-lea/ElectricGrid.jl #129: Status and need of example scripts

wallscheid, 2023-09-30:
I feel that some / most of the scripts within examples/scripts folder are not in a good shape:

- introduction statements per script are missing, that is, an explanation of each script's content and motivation
- inline documentation is mostly very sparse and erratic 
- some commands are commented out without further context
- some outputs after execution are irritating (e.g., env_eampe.jl outputs something regarding RL settings although it was never configured for RL-based control)  

Considering that some of these scripts have been not touched for months and that there seem also content overlaps to the notebooks from the examples folder I would like to ask the core devs to critically go through the scripts and either delete those which them not having added value or improve those which should be kept for documentation / tutorial purpose

Webbah, 2023-10-04:
All old scripts have been deleted, rest tidied up and documented. 
please comment if further need of adjustments or close if ok

Webbah, 2023-10-05:
Fine regarding last editorial checks

closed 2023-10-05 by Webbah