upb-lea/ElectricGrid.jl #130: Structure of example notebooks subfolder

wallscheid, 2023-09-30:
We have in total 18 notebooks, however, the subfolder's readme only introduces 10. Please add the missing notebooks to the readme.

Furthermore, it is irritating and inconsistent that some of the notebooks start with a number (while the numer '1' is even used twice). Please get rid of the numbers at the beginning of the notebooks' names. If absolutely necessary, you could add an '...._part01.ipynb' at the end of the relevant notebooks. 

Similarly, it remains unclear to me what the wording 'DEMO' should indicate as part of some notebooks' names? If I am going through example notebooks, I would automatically expect that they demonstrate something to me as a user, so why do we need that trivial 'DEMO' suffix? 

Further general details:

- 1_Auxiliaries_OU_process notebooks has an introductory table with one item, is that necessary?
- The RenderHookResults() did not succesfully render something, i.e., they calculated for ever (more than 15 minutes) without any grahical output, although the Julia env was set up and configured according to the GitHub instructions (actually, the script examples' graphical outputs worked well) . I guess that an Jupyter-related issue associated with my local machine setup, however, I just report here that following the installation routine of ElectricGrid.jl package it does not render out of box.

Webbah, 2023-10-04:
- readme is updated
- notebooks renamed

Webbah, 2023-10-05:
- Issue seems to be related to PlotlyJS connected to Julia 1.9.3. which is not part of our developed tool
-  working with Julia 1.8.5. seemed to fix the issue being able to produce the ElectricGrid.jl graphical notebook outputs

closed 2023-10-05 by Webbah